Bodmer/TFT_eFEX #3: Graph and meters

sorriso93, 2019-04-01:
I know you are already planning this but just to inform you some code is already available maybe you can easily integrate it (I tested it and it is almost ok with your library).

Found on github various code maybe you can improve it
many thanks

ringmeter below

`/*
 An example showing 'ring' analogue meters on a 2.2" ILI9341 TFT 240x320
 colour screen
 
 The meter graphic function works best with slowly changing values without
 large rapid changes such as environmental parameters (temperature etc)

 This example uses the hardware SPI and a board based on the ATmega328
 such as an UNO.  If using a Mega or outher micocontroller then comment out
 the following line:
 #define F_AS_T
 in  the "Adafruit_ILI9341_FAST.h" found within the Adafruit_ILI9341_AS
 library folder. Change pins for other Arduinos to the hardware SPI pins.

 Needs Fonts 2, and 4 (also Font 6 if using a large size meter)

 Alan Senior 18/3/2015
 */

// These are the connections from the UNO to the display
#define sclk 13  // Don't change
#define mosi 11  // Don't change
#define cs   10  // If cs and dc pin allocations are changed then 
#define dc   9   // comment out #define F_AS_T line in "Adafruit_ILI9341_FAST.h"
// (which is inside "Adafriit_ILI9341_AS" library)

#define rst  7  // Can alternatively connect this to the Arduino reset

// Meter colour schemes
#define RED2RED 0
#define GREEN2GREEN 1
#define BLUE2BLUE 2
#define BLUE2RED 3
#define GREEN2RED 4
#define RED2GREEN 5

#include <Adafruit_GFX_AS.h>     // Core graphics library
#include <Adafruit_ILI9341_AS.h> // Fast hardware-specific library
#include <SPI.h>

#define ILI9341_GREY 0x2104 // Dark grey 16 bit colour

Adafruit_ILI9341_AS tft = Adafruit_ILI9341_AS(cs, dc, rst); // Invoke custom library

uint32_t runTime = -99999;       // time for next update

int reading = 0; // Value to be displayed
int d = 0; // Variable used for the sinewave test waveform

void setup(void) {
  tft.init();

  tft.setRotation(3);

  tft.fillScreen(ILI9341_BLACK);
}

void loop() {
  if (millis() - runTime >= 2000L) { // Execute every 2s
    runTime = millis();

    // Test with a slowly changing value from a Sine function
    d += 5; if (d >= 360) d = 0;
[... 439 lines cut ...]
    //now draw the labels
    d.setTextSize(0);
    d.setTextColor(tcolor, bcolor);
    d.setCursor(gx , gy - h - 30);
    d.println(title);
    d.setTextSize(0);
    d.setTextColor(acolor, bcolor);
    d.setCursor(gx , gy + 20);
    d.println(xlabel);
    d.setTextSize(0);
    d.setTextColor(acolor, bcolor);
    d.setCursor(gx - 30, gy - h - 10);
    d.println(ylabel);
  }
  //graph drawn now plot the data
  // the entire plotting code are these few lines...
  // recall that ox and oy are initialized as static above
  x =  (x - xlo) * ( w) / (xhi - xlo) + gx;
  y =  (y - ylo) * (gy - h - gy) / (yhi - ylo) + gy;
  d.drawLine(ox2, oy2, x, y, pcolor);
  d.drawLine(ox2, oy2 + 1, x, y + 1, pcolor);
  d.drawLine(ox2, oy2 - 1, x, y - 1, pcolor);
  ox2 = x;
  oy2= y;
}`

Bodmer, 2019-04-01:
Thanks for the info. I will have a look.

closed 2019-04-01 by Bodmer

Alexandre2003, 2020-03-29:
Hello friend, well I saw that you passed these information to Bodmer look I really need to create vertical and horizontal bars I tried with the code that you described there above but it didn't work I'm a beginner and it was difficult for me I asked Bodmer for help but I went iguinorado kkkk if you can help me or give me a semi finished condition i will be forever grateful to you sorry for the inconvenience ok i'm from brazil

Att Alexandre Guaita

https://github.com/Bodmer/TFT_eFEX/issues/3#issue-427942037

Bodmer, 2020-03-29:
[Look here.](https://forum.arduino.cc/index.php?topic=671859.msg4534433#msg4534433)

sorriso93, 2020-03-30:
Ciao, you can also look into [this example ](https://create.arduino.cc/projecthub/LightPro/tft-graphing-bar-charts-185436)