Bodmer/TFT_eFEX #4: ESP32 drawjpg from array not working

nikisalli, 2019-04-21:
hi :)
i'm having an issue with both drawArrayJpeg() functions. 
it looks like all the chunks of the image are rendered in the exact same spot and i couldn't figure out why because with your jpeg decoder library it works so my array can't be wrong.
[image]
maybe incorrect offset calculation for the chunks?
this is the code i'm using. 
you can see i included both libraries and at the last line i called the jpg displayer.

```
#include <WiFi.h>
#include <WiFiMulti.h>
#include <HTTPClient.h>
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
#include <TFT_eFEX.h>   
TFT_eFEX  fex = TFT_eFEX(&tft);

#define USE_SERIAL Serial

WiFiMulti WiFiMulti;
int pic = 1;

uint8_t PicArray[80750] = {0}; //Array for picture make sure under 15K

void setup() {
  pinMode(23, OUTPUT);
  digitalWrite(23, LOW);
  pinMode(27, INPUT);
  
  USE_SERIAL.begin(115200);
  
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(ILI9341_GREEN);
  tft.setCursor(15, 30);
  tft.setTextSize(2);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  digitalWrite(23, HIGH);
  
  //###################################################################

  WiFiMulti.addAP("Vodafone-A48216342", "12344321");
  //####################################################################

}

void loop() {
  if (pic > 8)
  {
    pic = 1;
  }

  //tft.fillScreen(TFT_BLACK);

  tft.fillRect(11, 26, 240, 24, 0x0000);
  tft.drawRect(11, 26, 240, 24, ILI9341_RED);
  tft.setTextColor(ILI9341_GREEN);
  tft.setCursor(15, 30);
  tft.setTextSize(2);
  tft.print("Downloading Pic:" + String(pic));
  USE_SERIAL.println("Downloading Pic:" + String(pic));
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    GetPic();
  }
  pic++;
  delay(10000);
}

void GetPic() {
  HTTPClient http;
  long totalSize = 0;
  boolean chone = 1;
  // configure server and url update based on your URL
  http.begin("http://www.meteosatonline.it/riduce/nuova/ultima.jpg");  //update based on your URL
  //http.begin("http://10.16.10.11:8980/tempgr/ac/BaboonL.jpg");
  // start connection and send HTTP header
  long httpCode = http.GET();
  if (httpCode > 0) {
    // HTTP header has been send and Server response header has been handled
    USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

    // file found at server
    if (httpCode == HTTP_CODE_OK) {

      // get lenght of document (is -1 when Server sends no Content-Length header)
      long len = http.getSize();

      // create buffer for read
      uint8_t buff[2048] = { 0 };

      // get tcp stream
      WiFiClient * stream = http.getStreamPtr();

      // read all data from server
      while (http.connected() && (len > 0 || len == -1)) {
        size_t size = stream->available();
        if (size) {

          int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
          memcpy (PicArray + totalSize, buff, c);
          totalSize = totalSize + c;

          if (len > 0) {
            len -= c;
          }
        }

        yield();
      }
      USE_SERIAL.print("[HTTP] connection closed or file end.\n");

    }
  } else {
    USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
  }

  http.end();

  USE_SERIAL.println("TotalS:" + String(totalSize));

  USE_SERIAL.print("First 10 Bytes: ");
  for (int ipt = 0; ipt < 11; ipt++) {
    USE_SERIAL.print(PicArray[ipt], HEX);
    USE_SERIAL.print(",");
  }
  USE_SERIAL.print("\nLast 10 Bytes : ");
  for (int ipt = 10; ipt >= 0; ipt--) {
    USE_SERIAL.print(PicArray[totalSize - ipt], HEX);
    USE_SERIAL.print(",");
  }
  USE_SERIAL.println("");
  uint8_t scale = (uint8_t)JPEG_DIV_2;
  fex.drawJpg(PicArray, sizeof(PicArray), 0, 0);
  //drawArrayJpeg(PicArray, totalSize + 10, 0, 0);
  //drawArrayJpeg(PicArray,sizeof(PicArray), 0, 0);
}
```

Bodmer, 2019-05-08:
I tried your code and it worked fine fetching small images (<40kbytes) from a server.

The image you are trying to load is far too big for the ESP32 RAM so I am surprised it does not crash. Try with smaller images.

closed 2019-05-08 by Bodmer

nikisalli, 2019-05-08:
Yes, in the meantime i solved the problem and found smaller size images from a different server so that they can fit in ram. I was able to display images up to 80kb in size. awesome!