Bodmer/TFT_eFEX #9: Landscape images do not display correctly with function drawJpgFile()

geprc, 2019-10-24:
ESP32 dev module with ST7789 display connected to hardware SPI.

When trying to display a SPIFFS jpg image which is 135x135 landscape orientation the image displays smaller with a block of background colour filling the rest of the screen.

I've modified the example sketch like this:
```c++
void loop() {
       tft.fillScreen(random(0xFFFF));
       tft.setRotation(1);
       fex.drawJpgFile(SPIFFS, "/image.jpg", 0, 0);
}
```
Just like this:
[image]
[image]

Bodmer, 2019-10-24:
It looks your screen has an offset of sbout 40 pixels. A quick fix is to add this offset in your sketch. Instead of 0,0 cordinates, try 40,40

geprc, 2019-10-25:
40， 40 will be this 
[image]
and 40， 70 will be this 
[image]

geprc, 2019-10-25:
and if i use function drawBmp(), it will be ok,it seems that function drawJpgFile() is not adaptation with st7789,or pixel 240*135

Bodmer, 2019-10-26:
I have marked this as a bug. Thanks for raising the issue. The display you are using has some offsets so it appears that these are not being used when rendering a jpeg file.

You could try this [new jpeg library](https://github.com/Bodmer/TJpg_Decoder).

I am not in a position to investigate the bug at the moment but will have a look in a few days time.

Bodmer, 2019-10-31:
I have tested it with the new [Tjpg_Decoder library](https://github.com/Bodmer/TJpg_Decoder) here and this example works: 
[SPIFFS_Jpg_T1.zip](https://github.com/Bodmer/TFT_eFEX/files/3796193/SPIFFS_Jpg_T1.zip)

Using the Tjpg_Decoder library is therefore an option for you until I can track down the bug.

Bodmer, 2019-11-29:
I think I will be stripping out the ESP32 related jpeg code as it has some rather awkward API, and update the examples to use the [TJpg_Decoder library](https://github.com/Bodmer/TJpg_Decoder) as that uses a later version of the decoder engine.

closed 2020-03-10 by geprc