Bodmer/TFT_eFEX #15: DrawBmp from Array

alijani1, 2019-12-02:
please add ability to drawbmp from an array  (16 bit 565) or (8bit)

Bodmer, 2019-12-02:
There is an example here that draw 16 bit raw bitmap image files. Os this waht you are asking for?

alijani1, 2019-12-02:
there is an interface to DrawJPG from array or a file, but there is no implementation of DrawBmp from array.  So developers have to create their own implementation using DrawPixel one at a time if they have the BMP already in array.   Since you have a DrawJPG and you can pass an array, in the same if you can implement DrawBmp  version where you can also pass array instead of  file name, it would make it more complete and eliminate developers from creating their own draw from array functions.

Bodmer, 2019-12-03:
I missed the link. [Here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_Flash_Bitmap/TFT_Flash_Bitmap.ino).
See comment on how to create the bitmap.
If you mean store a windows format bitmap then this would need a method to copy the BMP file into an array to be identified.  Note that the BMP format would be very wasteful of memory due to the header size and 3 bytes per pixel and would probably be of interest to very few users.

alijani1, 2019-12-03:
What I meant was ability to send 565 or 332 image data stored in array to sprite or screen.  I did not realize you had implemented PushImage when I posted this so I guess its similar except PushImage does not support 8 bit images (i think).   I guess no action needed and we can close out.  Thanks for your replies.  Love the library.

Bodmer, 2019-12-03:
There is a pushImage for [8 bit colours and 1 bit](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L760-L762). If the image pointer is 16 bit, this uses the 16 bit colour pushImage, if the pointer is 8 bit the 8bit colours are assumed.

8 bit pointers are also used for 1 bit colour images, so the last parameter selects 1 bit colours if the last parameter is set false, the default is true is the last parameter is missing.

Colour format for 8 bits is 332 - RGB, so blue is only 2 bits, red and green 3 bits.

closed 2019-12-03 by Bodmer