Bodmer/TFT_eFEX #18: TJpgDec.drawJpg vs fex.drawJpg

robotzero1, 2020-04-17:
Hi, I've connected a ST7735 to an ESP32-CAM for a live video view.

Both of the following work but I'm not sure of the advantages and disadvantages of each
```
fex.drawJpg((const uint8_t*)fb->buf, fb->len, 0, 0);
TJpgDec.drawJpg(0, 0, (const uint8_t*)fb->buf, fb->len);
```
The eFex library has more features and offers native ESP32 decoding so for my use I guess this is the one to use.

Bodmer, 2020-04-17:
Both methods use the same underlying jpeg decoder software and so the performance is likely to be identical. TJpg_Decoder uses a slightly newer version of the decoder software but that probably makes no difference. TJpg_Decoder is a generic library so can be used with other processors such as STM32 or ESP8266.

closed 2020-04-17 by Bodmer

summerfind, 2022-10-25:
Hi, Bodmer,
how to use TJpg_Decoder  to draw a Jpeg file on ESP32 SPIFFS?
Thanks
Summer