Bodmer/TFT_eFEX #19: Platformio compile time issues

jordancrubin, 2020-04-17:
This method all seems to work if TFT_eFEX is not used

if platformio.ini is configured as shown below, the following compile errors are noted below.  This is only the case when the first param USER_SETUP_LOADED=1 is enabled, when remarked out, it compiles fine. This idea taken from the tft_espi guidance at

https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/PlatformIO/Configuring%20options.txt

The configs shown below would be for a TTGO T displayv1

platformio.ini 

build_flags =
-Os
-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
-DUSER_SETUP_LOADED=1
-DST7789_DRIVER=1
-DTFT_WIDTH=135
-DTFT_HEIGHT=240
-DCGRAM_OFFSET=1
-DTFT_MOSI=19
-DTFT_SCLK=18
-DTFT_CS=5
-DTFT_DC=16
-DTFT_RST=23
-DTFT_BL=4
-DTFT_BACKLIGHT_ON=HIGH
-DLOAD_GLCD=1
-DLOAD_FONT2=1
-DLOAD_FONT4=1
-DLOAD_FONT6=1
-DLOAD_FONT7=1
-DLOAD_FONT8=1
-DLOAD_GFXFF=1
-DLOAD_SMOOTH_FONT=1
-DSPI_FREQUENCY=40000000
-DSPI_READ_FREQUENCY=6000000

Compiling .pio/build/ttgo-lora32-v1/lib3aa/TFT_eFEX_ID6164/TFT_eFEX.cpp.o
Archiving .pio/build/ttgo-lora32-v1/lib2ae/libSD.a
Indexing .pio/build/ttgo-lora32-v1/lib2ae/libSD.a
Archiving .pio/build/ttgo-lora32-v1/libFrameworkArduinoVariant.a
Indexing .pio/build/ttgo-lora32-v1/libFrameworkArduinoVariant.a
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/FunctionalInterrupt.cpp.o
Archiving .pio/build/ttgo-lora32-v1/lib67c/libJPEGDecoder_ID1467.a
Indexing .pio/build/ttgo-lora32-v1/lib67c/libJPEGDecoder_ID1467.a
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/ttgo-lora32-v1/FrameworkArduino/MD5Builder.cpp.o
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp: In member function 'void TFT_eFEX::drawJpeg(String, int16_t, int16_t, TFT_eSprite*)':
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:287:13: warning: variable 'disp_w' set but not used [-Wunused-but-set-variable]
     int32_t disp_w = _tft->width();
             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp: In member function 'void TFT_eFEX::drawJpeg(const uint8_t*, uint32_t, int16_t, int16_t)':
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:391:13: warning: unused variable 'disp_w' [-Wunused-variable]
     int32_t disp_w = _tft->width();
             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp: In member function 'void TFT_eFEX::drawStringRTL(const char*, int32_t*, int32_t*)':
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:763:13: error: 'class TFT_eSPI' has no member named 'fontLoaded'
   if (_tft->fontLoaded)
             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:774:26: error: 'class TFT_eSPI' has no member named 'getUnicodeIndex'
       bool found = _tft->getUnicodeIndex(uniCode, &gNum);
                          ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:778:22: error: 'class TFT_eSPI' has no member named 'gFont'
         poX -= _tft->gFont.spaceWidth;
                      ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:783:22: error: 'class TFT_eSPI' has no member named 'gFont'
         poY += _tft->gFont.yAdvance;
                      ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:788:24: error: 'class TFT_eSPI' has no member named 'gWidth'
         poX -= ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                        ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:788:45: error: 'class TFT_eSPI' has no member named 'gdX'
         poX -= ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:792:41: error: 'class TFT_eSPI' has no member named 'gWidth'
           poX = _tft->width() - ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                         ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:792:62: error: 'class TFT_eSPI' has no member named 'gdX'
           poX = _tft->width() - ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                                              ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:793:24: error: 'class TFT_eSPI' has no member named 'gFont'
           poY += _tft->gFont.yAdvance;
                        ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:798:15: error: 'class TFT_eSPI' has no member named 'drawGlyph'
         _tft->drawGlyph(uniCode);
               ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:767:13: warning: unused variable 'cy' [-Wunused-variable]
     int16_t cy = _tft->getCursorY();
             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp: In member function 'void TFT_eFEX::drawStringLTR(const char*, int32_t*, int32_t*)':
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:838:13: error: 'class TFT_eSPI' has no member named 'fontLoaded'
   if (_tft->fontLoaded)
             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:855:26: error: 'class TFT_eSPI' has no member named 'getUnicodeIndex'
       bool found = _tft->getUnicodeIndex(pointCodes[index], &gNum);
                          ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:859:22: error: 'class TFT_eSPI' has no member named 'gFont'
         poX -= _tft->gFont.spaceWidth;
                      ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:864:22: error: 'class TFT_eSPI' has no member named 'gFont'
         poY += _tft->gFont.yAdvance;
                      ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:869:24: error: 'class TFT_eSPI' has no member named 'gWidth'
         poX -= ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                        ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:869:45: error: 'class TFT_eSPI' has no member named 'gdX'
         poX -= ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                             ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:873:41: error: 'class TFT_eSPI' has no member named 'gWidth'
           poX = _tft->width() - ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                         ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:873:62: error: 'class TFT_eSPI' has no member named 'gdX'
           poX = _tft->width() - ( _tft->gWidth[gNum] + _tft->gdX[gNum] );
                                                              ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:874:24: error: 'class TFT_eSPI' has no member named 'gFont'
           poY += _tft->gFont.yAdvance;
                        ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:879:15: error: 'class TFT_eSPI' has no member named 'drawGlyph'
         _tft->drawGlyph(pointCodes[index]);
               ^
/Users/jord/.platformio/lib/TFT_eFEX_ID6164/TFT_eFEX.cpp:842:13: warning: unused variable 'cy' [-Wunused-variable]
     int16_t cy = _tft->getCursorY();
             ^
*** [.pio/build/ttgo-lora32-v1/lib3aa/TFT_eFEX_ID6164/TFT_eFEX.cpp.o] Error 1

Bodmer, 2020-04-18:
I use the Arduino IDE and it runs fine. The errors may be down to the way the PlatformIO environment has loaded the library. The functions that are in error are within the TFT_eSPI library so that must be loaded. Unfortunately I am unable to help further due to unfamiliarity with the PlatformIO IDE.

Bodmer, 2020-04-18:
Use #define SMOOTH_FONT not LOAD_SMOOTH_FONT

closed 2020-04-18 by Bodmer

jordancrubin, 2020-04-19:
ok, I thought becuase you had documentation for platformio in the code that it was supported.  I will note that this runs fine in Arduino IDE, I was trying to port it over using this method without modifying to use setup 25 specifically and globally.

Bodmer, 2020-04-19:
I think it will work if you correct your setup to be:
-DSMOOTH_FONT=1

jordancrubin, 2020-04-20:
You were correct!! I had messed up that one parameter, its correction allowed the program to compile and function without issue. Much thanks. If I may, I feel as though the only thing missing for platformio.ini with regard to your library is one build parameter, such as your user_setup_select.h, instead of copying in all of the define attribs manually.

for example:
lib_deps = tft_eFEX , tft_eSPI

build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  **-DUSER_SETUP=25** // TTGO TDISPLAY FOR EXAMPLE