Bodmer/TFT_eFEX #27: drawSdJpeg doesn't work for some jpeg images

PLSousa, 2020-11-11:
Hello Bodmer,
Many thanks for this development.

I have used your library to display jpeg images stored in a SD card or in SPIFFS. It has worked successfully for many jpeg files, but not for these ones (cover.jpg, cover2.jpg, cover3.jpg).
Do you have any idea why ?

Basically, I use an ESP32 running ESP32_SDcard_jpeg.ino or Jpeg_ESP32.ino. 

`ESP32 Jpeg decoder test!`
`Initialisation done.`
`Listing SPIFFS files:`
`=================================================`
`File name                              Size`
`=================================================`
`/cover2.jpg                           14082 bytes`
`/EagleEye.jpg                         37755 bytes`
`/cover3.jpg                           27765 bytes`
`/cover.jpg                            18487 bytes`
`=================================================`

`  fex.drawJpgFile(SPIFFS, "/EagleEye.jpg", 0, 0);   // it displays an image`

`  fex.drawJpgFile(SPIFFS, "/cover.jpg", 0, 0);      // nothing is displayed in the TFT 240x240_ST7789`

[image]
[image]
[image]

Bodmer, 2020-11-13:
jpeg images can be encoded in different ways and not all are supported. You may find that loading the images into a program like Paint, and then saving them as a jpeg will fix it.

PLSousa, 2020-11-13:
thank you Bodmer. But I would like to download Jpeg images from the web (
www.fip.fr) and display them, in real time. It is for a webradio project
using ESP32.
"cover1.jpg is an example of downloaded image.

Bodmer, 2020-11-13:
Image analysis indicates the jpegs you cannot display are in "Progressive" format and this is not supported by the JPEGDecoder library or TJpg_Decoder libraries. So I have no solution for you :-(

closed 2020-11-13 by Bodmer

PLSousa, 2020-11-13:
thanks anyway