Bodmer/TFT_eFEX #28: TFT_eFEX and Sprites

eb4cuv, 2020-12-02:
Hi bodmer

I would like to load a jpg image into a sprite and then do a pushSprite on the screen.

It doesn't work and I don't know what I'm doing wrong.

#include <TFT_eSPI.h> // Hardware-specific library
#include <TFT_eFEX.h> // Extension graphics functions library

TFT_eSPI tft = TFT_eSPI();

TFT_eSprite smeter = TFT_eSprite(&tft);

TFT_eFEX fex_presentacion = TFT_eFEX(&tft); 
TFT_eFEX fex_smeter = TFT_eFEX(&smeter);

// ***in setup ***
smeter.setColorDepth(8);  
smeter.createSprite(480, 190);
smeter.fillSprite(TFT_BLACK);

SPIFFS.begin();
fex_smeter.drawJpgFile(SPIFFS, "/smeter.jpg", 0, 0);

smeter.pushSprite(0,0);

without actually executing smeter.pushSprite (0,0); the image is displayed on the screen and it is not what I want. I want to load it into the sprite, modify it and then present it on screen with pushSprite.

Thank you. These libraries are a great contribution. Thank you very much for your work.

Bodmer, 2020-12-03:
Have a look at the example Rotated_Sprite_3. A jpeg is loaded into the sprite in the setup()
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Sprite/Rotated_Sprite_3

eb4cuv, 2020-12-04:
Thanks. It's already working.

closed 2020-12-17 by Bodmer