Bodmer/TFT_eFEX #29: Library or Compile problems

bryanpedersen, 2020-12-17:
Probably a basic problem, I am more interested in the coding than the environment and this does seem like an environment problem.

I have a program for the AI-Thinker-ESP32-CAM delivering still images by WiFi and also have a TFT display connected. The program uses TFT_eSPI and can capture frames, view over WiFi and I have a basic Hello World appearing on the TFT display. However, simply adding the include for the TFT_eFEX library causes lots of compile errors 'class TFT_eSPI' has no member named...
No other change in the code. All libraries, as well as JPEGDecoder (although not included at this stage) have been installed via downloaded ZIP file.
Any suggestions appreciated.

bryanpedersen, 2020-12-17:
Additional information. I am using Arduino IDE 1.8.13
Not all members of TFT_eSPI cause an error, only fontLoaded, getUnicodeIndex, gFont, gWidth, gdX and drawGlyph.
Thanks.

bryanpedersen, 2020-12-17:
Solved, sort of. In the user_setup.h file there is a suggestion to save memory by commenting out #define SMOOTH_FONT but if you do that it causes these compile issues. I guess there is some way around that, but I am moving again anyway.

closed 2020-12-17 by bryanpedersen

Bodmer, 2020-12-17:
I have updated the library so these compile errors will not occur. Thanks for reporting this.