Bodmer/TFT_eFEX #30: Arabic Support to RTL

MShokry, 2021-01-16:
The arabic letters is connected letter ex
ببب in utf8 all are "ب" letter
how to solve
1. Find the position of the letter related to prev. and next letters 
2. replace the code in ('<isolated>', '<initial>', '<medial>', '<final>')

Bodmer, 2021-01-16:
Merged, thanks. I have not tested the changes, so please standby incase there are any issues raised.

SiavashSkynet, 2021-09-15:
Hi Bodmer and Mohammad

Thank you for this very stunning repository.
I am using a LCD ILI9488 and ESP32 to print an Arabic sentence on the screen.
Every things work good and I did not see any strange thing in process. I used drawStringRTLAr example created by Mr Shokri.
The issue is that when I use `drawStringRTLAr(const String &input)` in order to print an Arabic sentence in my code ,  the printing of my input is character by character with delay and it is visible bye eye.
I mean Is there any way to print Arabic or Farsi words on the TFT like English words?