Bodmer/TFT_eFEX #33: Bug? TFT_eFEX ignores screen offset set on TFT_eSPI

zaraken, 2021-02-05:
I have roughly the following setup

I have an ESP32 TTGO with ST7789_DRIVER 
`CGRAM_OFFSET` is defined  
my code is:
```
TFT_eSPI tft = TFT_eSPI();
TFT_eFEX tft_x = TFT_eFEX(&tft);

tft.begin();
// tft.setOrientation(1); // setting the orientation on tft object

tft_x.drawJpg(img_buff, buff_size, 0, 0); // ESP32 optimized 
```
The problem is (was) that the TFT_eFEX::drawJpg was not applying the necessary offset to the picture and as a result the picture was cropped/not centered. And in fact it looked like it was ignoring configurations that are applied to the TFT_eSPI object such as the setOrientation. 

I traced it to this line https://github.com/Bodmer/TFT_eFEX/blob/2b127258d8ee35122074a874aa1354c597e73a71/TFT_eFEX.cpp#L1488

where jpeg.tft is assigned the extended TFT_eFEX object (`this` same as `tft_x` from my code above) which is itself also a TFT_eSPI object but not the original one on which the configurations are applied. The original object (`tft` from my code) is stored in `tft_x` during TFT_eFEX instantiation here https://github.com/Bodmer/TFT_eFEX/blob/2b127258d8ee35122074a874aa1354c597e73a71/TFT_eFEX.cpp#L18

So I changed the line 1488 to 
`jpeg.tft = _tft;` 
and that fixed this particular problem that I had. But I'm not sure if this would create other problems. Is this a bug that I found, or is this just a workaround for my lack of understanding on how to use the library properly?

Alternatively could I instead apply configurations on the extended object directly like so
```
TFT_eSPI tft = TFT_eSPI();
TFT_eFEX tft_x = TFT_eFEX(&tft);

tft_x.begin();
// tft_x.setOrientation(1); // setting the orientation on tft object

tft_x.drawJpg(img_buff, buff_size, 0, 0); // ESP32 optimized 
```

And if that is possible do I need to instantiate TFT_eSPI at all ? Can't I just use TFT_eFEX?

zaraken, 2021-02-05:
One clarification, my code was based on the library's examples https://github.com/Bodmer/TFT_eFEX/blob/master/examples/Jpeg_ESP32/Jpeg_ESP32.ino

jsmestad, 2021-05-09:
@zaraken thanks for the fix. I was having the same issue 👍