Bodmer/TFT_eFEX #35: GC9A01 Rotated_Sprite_3.ino pixels of random colors- compression error?

flykarlos, 2021-02-25:
First of all:
Thanks @Bodmer  for this wonderful library and tremendous amount of work You are putting into maintenance of it!

I started using Your library to drive GC9A01 display. Everything works fine, except rotation of sprite with jpeg plotted on it. I thought that there is error in my code, so I reverted back to examples and flashed unchanged Rotated_Sprite_3.ino example. Result is that some image is plotted on the screen. There is some similarity to image stored in data folder, however looks like each end every pixel has random color. Just like some kind of compression error. Have a look:

[image]
[image]

Any help will be appreciated.
Processor: ESP32 ESP-WROOM-32D
environment: MacOS Mojave, Arduino IDE 1.8.13
Of course I am using ESP32FS tool to upload sketch data as described in example.
Thanks!

flykarlos, 2021-02-25:
Thanks for fast reply!

Your workaround works perfectly. This problem is now gone. But now I see, that it was not an issue that I experienced in my code. 
I am proggraming compas. Dial is 240x240 px JPEG stored as array and projected to sprite. Then sprite is rotated by known angle (pivot point is screen center). 
Problem is that wen I use `spr.pushRotated(30); // fixed angle for test purposes` image is distorted. 
when I just `spr.pushSprite(0, 0);` image looks perfectly fine.

[image]
[image]
original jpeg:
[image]

Do You have any idea what may I do wrong?
thanks!

flykarlos, 2021-02-25:
Looks like I found an answer.
When I use `spr.pushRotated(angle);` function I used only angle parameter, leaving transparency setting default. I thought that if I leave second parameter empty, library will by default ignore any transparency and make all colors solid. But that's wrong. Looks like library default is set to TFT_WHITE, and this color becomes transparent  making black pixels from below sprite being visible. Workaround is to manually chose color to be transparent and make sure this color is not used in rotated .JPG file. In my case I chose `spr.pushRotated(angle, TFT_PINK);`  and image looks now far better after rotation.

Bodmer, 2021-02-25:
There was a bug that was only corrected yesterday that may be the problem. An update of Sprite.h and Sprite.cpp should fix that.

The rotation method used is quite basic, it selects the nearest pixel, A better method would be to super-sampe the image (5 x 5 samples per pixel) and average the colour.  This is on the todo list.

Bodmer, 2021-02-25:
I tried the compass rotation and it works quite well but is rathr slow for a whole scree. You could try the dial rotating in a viewport area.  This sketch shows how it works, you need a copy of your image file called compass.jpg in SPIFFS.
```
//====================================================================================
//                                  Libraries
//====================================================================================
// Call up the SPIFFS FLASH filing system, this is part of the ESP Core
#define FS_NO_GLOBALS
#include <FS.h>

#ifdef ESP32
#include "SPIFFS.h" // Needed for ESP32 only
#endif

// https://github.com/Bodmer/TFT_eSPI
#include <TFT_eSPI.h>                 // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();            // Invoke custom library
TFT_eSprite spr = TFT_eSprite(&tft);  // Create Sprite object "spr" with pointer to "tft" object

// https://github.com/Bodmer/TFT_eFEX
#include <TFT_eFEX.h>                 // Include the function extension library
TFT_eFEX fex = TFT_eFEX(&tft);        // Create TFT_eFX object "fex" with pointer to "tft" object

//====================================================================================
//                                    Setup
//====================================================================================
void setup()
{
  Serial.begin(250000); // Used for messages

  tft.begin();
  tft.setRotation(0);  // 0 & 2 Portrait. 1 & 3 landscape
  tft.fillScreen(TFT_BLACK);
  spr.setColorDepth(8);
  // Create a sprite to hold the jpeg (or part of it)
  spr.createSprite(240, 240);

  // Initialise SPIFFS
  if (!SPIFFS.begin()) {
    Serial.println("SPIFFS initialisation failed!");
    while (1) yield(); // Stay here twiddling thumbs waiting
  }
  Serial.println("\r\nInitialisation done.\r\n");

  // Lists the files so you can see what is in the SPIFFS
  fex.listSPIFFS();

  // Note the / before the SPIFFS file name must be present, this means the file is in
  // the root directory of the SPIFFS, e.g. "/tiger.jpg" for a file called "tiger.jpg"

  // Send jpeg info to serial port
  fex.jpegInfo("/compass.jpg");

  // Draw jpeg iamge in Sprite spr at 0,0
  fex.drawJpeg("/compass.jpg", 0 , 0, &spr);

  tft.setSwapBytes(true);
}

//====================================================================================
//                                    Loop
//====================================================================================
void loop()
{

  tft.fillScreen(random(0xFFFF));

  // Set the TFT pivot point to the centre of the screen
  tft.setPivot(tft.width() / 2, tft.height() / 2 - 2);
  // Set Sprite pivot point to centre of Sprite
  spr.setPivot(spr.width() / 2, spr.height() / 2);

  tft.setViewport(40, 0, 160, 60);

  // Push copies of Sprite rotated through increasing angles 0-360 degrees
  // with 1 degree increments
  for (int16_t angle = 0; angle <= 360; angle += 1) {
    spr.pushRotated(angle);
    //delay(500);
  }

  tft.resetViewport(); // Must reset so pivots are set for full screen window

  delay(2000);
}
//====================================================================================
```

flykarlos, 2021-02-25:
@Bodmer, maybe little offtopic but please have a look on animated dial example and GC9A01 if You have it. When I use this example needle leaves black empty trail when changing position. I know that this example should be demonstrating opposite- how to copy part of screen before needle covers it and then "paste" it back when needle left this specific area.