Bodmer/TFT_eFEX #37: Image problems using TFT_eFEX library

stephensaid, 2021-03-13:
Thanks, Bodmer.

I've implemented TFT_eFEX.

I now have a problem with the images. The following image is not right.
[image]

It should look like:
[image]

I have other images with the same behaviour.

I've tried changing tft.setSwapBytes(true); to tft.setSwapBytes(false); but it has no effect.

Please help.

_Originally posted by @stephensaid in https://github.com/Bodmer/TFT_eSPI/issues/1018#issuecomment-789177553_

Bodmer, 2021-03-13:
@fotosettore the OP question is for TFT_eFEX, Rotated_Sprite_3 in an example of TFT_eSPI so please raise a new issue.

stephensaid, 2021-09-19:
Dear Bodmer, 

Hope you are doing well.

It has been a while but had other things to work on and temporarily abandoned this project. I appreciate that you have left this issue open since then.

I updated TFT_eSPI to 2.3.70 and am running TFT_eFEX 0.0.8.

Please view the source [here](https://github.com/stephensaid/JuneRadio/releases/tag/0.1).

This is very much under construction and I am learning as I go.

As soon as you run the code, the first thing you will see is the splash screen which I have posited in my first post. There are other icons that have the same issue. You might want to replace config_sample.txt with config.txt and edit SSID and Password. Be sure to upload config.txt and most importantly the images and fonts to SPIFFS.

Many thanks in advance.
Stephen

stephensaid, 2023-05-14:
Hi, 

Changing fex.drawJpeg to fex.drawJpgFile fixed this issue.

closed 2023-05-14 by stephensaid