Bodmer/TFT_eFEX #39: setSwapBytes(true) not working for drawJpeg

wezleysherman, 2021-10-01:
Hi Bodmer,

Thank you for creating this awesome library!

I seem to be having an issue loading JPEG images from SPIFFS on the ESP32 using a ST7735S driver. 
I'm able to load BMP's using the drawBmp() FEX method with sprite->setSwapBytes(true); and everything looks perfect. The issue is that BMP's are really large compared to JPEGs. When I switch to use drawJpeg(), the swap bytes parameter seems to be ignored.

I attempted setting this for my tft, fex, and sprite pointers with no avail. 

How can I fix this? And is there any more info you need from me on it?

I even attempted using the Jpeg_ESP32.ino example and it experienced the same issue.

My code:

```
sprite->createSprite(64, 64);
sprite->setSwapBytes(true);
fex->drawJpeg("sprite.jpg", 0, 0, sprite);
```

Again, drawBmp works perfectly with setSwapBytes(true);

wezleysherman, 2021-10-01:
Manually commenting out line 297 in TFT_eFEX.cpp fixed the issue.

Paulie92, 2021-11-04:
Thank you @wezleysherman, it helps me so much!

I just had to keep in mind, that the setSwapBytes(true); command have to be in code either.