Bodmer/TFT_eFEX #42: PNG filetype support

simkard69, 2021-12-14:
Hello,

Do we have any chance of seeing PNG file type support to draw from files located in SPIFFS and/or SD card directly ?

Thanks !

Bodmer, 2021-12-14:
There are a few example sketches here:
https://github.com/Bodmer/PNG_TEST_ONLY

I have not created a library yet but it is on the to do list.

Bodmer, 2022-01-04:
I came across a [PNG decoder library](https://github.com/bitbank2/PNGdec) that is quite good. It does need a fair bit of RAM though (40kbytes?). It can be loaded via the Arduino IDE library manager.

The examples provided with that library would need to be modified to work with TFT_eSPI.

I created a test sketch for TFT_eSPI that decodes a small image stored in LittleFS. This was tested on an RP2040 board but should work fine on an ESP32:
[decPNG_Minimal.zip](https://github.com/Bodmer/TFT_eFEX/files/7805058/decPNG_Minimal.zip)

Bodmer, 2022-02-26:
PNG examples have been added to TFT_eSPI [here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/PNG%20Images).

closed 2022-02-26 by Bodmer