Bodmer/TFT_eFEX #45: TFT_eFEX and srite

powersoft, 2023-02-18:
Hello,

I have try to implement this command:

// Draw a Jpeg to the TFT, or to a Sprite if a Sprite instance is included
void drawJpeg(String filename, int16_t xpos, int16_t ypos, TFT_eSprite *_spr = nullptr);

When I compile the code as enclosed got an error:

'class TFT_eSprite' has no member named 'drawJpgFile';

What is wrong with my code?

`#define FS_NO_GLOBALS
#include "FS.h"
#include <SD.h>

#ifdef ESP32
#include "SPIFFS.h" // Needed for ESP32 only
#endif

#include <SPI.h>

// https://github.com/Bodmer/TFT_eSPI
#include <TFT_eSPI.h>                 // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();            // Invoke custom library
TFT_eSprite spr = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

// https://github.com/Bodmer/TFT_eFEX
#include <TFT_eFEX.h>              // Include the extension graphics functions library
TFT_eFEX  fex = TFT_eFEX(&tft);    // Create TFT_eFX object "efx" with pointer to "tft" object

uint8_t sdSelectPin = 25;

void setup() {
  Serial.begin(250000);

  tft.begin();
  spr.createSprite(192,192);
  spr.fillScreen(TFT_BLACK);
  tft.setRotation(1);  // 0 & 2 Portrait. 1 & 3 landscape
  tft.fillScreen(TFT_BLACK);

  if (!SD.begin(sdSelectPin)) Serial.println("SD Card Mount Failed");
  else Serial.println("SD Card Mount");
  
  if (!SPIFFS.begin()) {
    Serial.println("SPIFFS initialisation failed!");
    while (1) yield(); // Stay here twiddling thumbs waiting
  }
  Serial.println("\r\nInitialisation done.");
  //fex.listSPIFFS(); // Lists the files so you can see what is in the SPIFFS

  spr.drawJpgFile(SD, "/moon/180.jpg", 0, 0);
}

void loop() {
  // put your main code here, to run repeatedly:
}`

Bodmer, 2023-02-18:
That function is only available to the original ESP32, not the C3, S2 or S3.

Which processor are you using?

Bodmer, 2023-02-18:
The  correct line is:
fex.drawJpgFile(SD, "/moon/180.jpg", 0, 0);

If you want to draw to a sprite then use this line at the start of the sketch:
```
TFT_eFEX fex = TFT_eFEX(&spr);
```
instead of:
```
TFT_eFEX fex = TFT_eFEX(&tft);
```

closed 2023-02-18 by Bodmer